florisla/stm32loader #6: Error while writing "black pill"

Civilduino, 2018-08-06:
Hello!

First thank you very much for updating the loader to Python 3+
I am new with the STM32 and I have been trying to upload simple blink program with this loader without success.

I have the "Black pill" https://hobbycomponents.com/st-micro-electronics/960-stm32f103-black-pill-development-board and I am running Python 3.6.6 installed from miniconda.

After running the loader I get this error:

myuser$ python stm32loader.py -V -f F1 -e -w -p /dev/tty.usbserial-A1012H88 blink_ardu.bin
Open port /dev/tty.usbserial-A1012H88, baud 115200
*** Get command
    Bootloader version: 0x22
    Available commands: 0x0, 0x1, 0x2, 0x11, 0x21, 0x31, 0x43, 0x63, 0x73, 0x82, 0x92
Bootloader version 22
*** GetID command
Chip id: 0x410 (STM32F10x Medium-density)
*** ReadMemory command
Device UID: 0036-066E-35375930-43185029
*** ReadMemory command
Flash size: 64 KiB
*** Erase memory command
    Erase memory done
Write 256 bytes at 0x8000000
*** Write memory command
    [256] bytes to write
Traceback (most recent call last):
  File "stm32loader.py", line 660, in <module>
    main()
  File "stm32loader.py", line 654, in main
    loader.perform_commands()
  File "stm32loader.py", line 580, in perform_commands
    self.bootloader.write_memory_data(self.configuration['address'], binary_data)
  File "stm32loader.py", line 368, in write_memory_data
    self.write_memory(address, data[offset:offset + 256])
  File "stm32loader.py", line 271, in write_memory
    self.serial.write(bytearray([data]))
TypeError: an integer is required

I don't know why is this problem while writing.

Thanks if you can help.

Civilduino, 2018-08-06:
I dont know the cause of the bug. I can give you more details of how I use the stm32loader.
1) First I installed it using pip install setup.py
2) Then I run it from a folder where I made a copy of the stm32loader.py file, i.e. using python stm32loader.py

As I said before, I have Python 3.6.6 and here is the list of installed packages:

# packages in environment at /Users/myuser/miniconda3:
#
# Name                    Version                   Build  Channel
alabaster                 0.7.11                   py36_0  
appnope                   0.1.0            py36hf537a9a_0  
asn1crypto                0.24.0                   py36_0  
astroid                   1.6.5                    py36_0  
babel                     2.6.0                    py36_0  
backcall                  0.1.0                    py36_0  
blas                      1.0                         mkl  
bleach                    2.1.3                    py36_0  
blosc                     1.14.0                        1    conda-forge
bzip2                     1.0.6                         1    conda-forge
ca-certificates           2018.03.07                    0  
certifi                   2018.4.16                py36_0  
cffi                      1.11.5           py36h342bebf_0  
chardet                   3.0.4            py36h96c241c_1  
cloudpickle               0.5.3                    py36_0  
conda                     4.5.9                    py36_0  
conda-env                 2.6.0                h36134e3_0  
cryptography              2.2.2            py36h1de35cc_0  
cycler                    0.10.0                     py_1    conda-forge
dbus                      1.13.2               h760590f_1  
decorator                 4.3.0                    py36_0  
docutils                  0.14             py36hbfde631_0  
entrypoints               0.2.3            py36hd81d71f_2  
expat                     2.2.5                hb8e80ba_0  
freetype                  2.8.1                         0    conda-forge
gettext                   0.19.8.1             h15daf44_3  
glib                      2.56.1               h35bc53a_0  
hdf5                      1.10.1                        2    conda-forge
html5lib                  1.0.1            py36h2f9c1c0_0  
icu                       58.2                 h4b95b61_1  
idna                      2.7                      py36_0  
imagesize                 1.0.0                    py36_0  
intel-openmp              2018.0.3                      0  
ipykernel                 4.8.2                    py36_0  
ipython                   6.4.0                    py36_0  
ipython_genutils          0.2.0            py36h241746c_0  
isort                     4.3.4                    py36_0  
jedi                      0.12.0                   py36_1  
jinja2                    2.10             py36hd36f9c5_0  
jpeg                      9b                   he5867d9_2  
jsonschema                2.6.0            py36hb385e00_0  
jupyter_client            5.2.3                    py36_0  
jupyter_core              4.4.0            py36h79cf704_0  
keyring                   13.2.1                   py36_0  
kiwisolver                1.0.1                    py36_1    conda-forge
lazy-object-proxy         1.3.1            py36h2fbbe47_0  
libcxx                    4.0.1                h579ed51_0  
libcxxabi                 4.0.1                hebd6815_0  
libedit                   3.1.20170329         hb402a30_2  
libffi                    3.2.1                h475c297_4  
libgfortran               3.0.1                h93005f0_2  
libiconv                  1.15                 hdd342a3_7  
libpng                    1.6.34               he12f830_0  
libsodium                 1.0.16               h3efe00b_0  
markupsafe                1.0              py36h3a1e703_1  
matplotlib                2.2.2                    py36_1    conda-forge
mccabe                    0.6.1            py36hdaeb55d_0  
mistune                   0.8.3            py36h1de35cc_1  
mkl                       2018.0.3                      1  
mkl_fft                   1.0.4            py36h5d10147_1  
mkl_random                1.0.1            py36h5d10147_1  
[... 45 lines cut ...]
setuptools                39.2.0                   py36_0  
simplegeneric             0.8.1                    py36_2  
sip                       4.19.8           py36h0a44026_0  
six                       1.11.0           py36h0e22d5e_1  
snowballstemmer           1.2.1            py36h6c7b616_0  
sphinx                    1.7.5                    py36_0  
sphinxcontrib             1.0              py36h9364dc8_1  
sphinxcontrib-websupport  1.1.0                    py36_1  
spyder                    3.3.0                    py36_0  
spyder-kernels            0.2.4                    py36_0  
sqlite                    3.24.0               ha441bb4_0  
testpath                  0.3.1            py36h625a49b_0  
tk                        8.6.7                h35a86e2_3  
tornado                   5.0.2                    py36_0  
traitlets                 4.3.2            py36h65bd3ce_0  
urllib3                   1.23                     py36_0  
wcwidth                   0.1.7            py36h8c6ec74_0  
webencodings              0.5.1            py36h3b9701d_1  
wheel                     0.31.1                   py36_0  
wrapt                     1.10.11          py36hc29e774_0  
xz                        5.2.4                h1de35cc_4  
yaml                      0.1.7                hc338f04_2  
youtube-dl                2018.8.4                  <pip>
zeromq                    4.2.5                h378b8a2_0  
zlib                      1.2.11               hf3cbc9b_2

greboide, 2018-08-08:
i can reproduce it here :+1: 

''''
Bootloader version 22
Chip id: 0x410 (STM32F10x Medium-density)
Supply -f [family] to see flash size and device UID, e.g: -f F1
Write 256 bytes at 0x8000000
Traceback (most recent call last):
  File "/home/glauber/bin/stm32loader", line 660, in <module>
    main()
  File "/home/glauber/bin/stm32loader", line 654, in main
    loader.perform_commands()
  File "/home/glauber/bin/stm32loader", line 580, in perform_commands
    self.bootloader.write_memory_data(self.configuration['address'], binary_data)
  File "/home/glauber/bin/stm32loader", line 368, in write_memory_data
    self.write_memory(address, data[offset:offset + 256])
  File "/home/glauber/bin/stm32loader", line 271, in write_memory
    self.serial.write(bytearray([data]))
TypeError: an integer is required
''''

florisla, 2018-08-08:
This bug is fixed in v0.3.3 (see #7 for the credits).

Please `pip install --upgrade stm32loader` and try again.