florisla/stm32loader #9: CRC Wrong in write_memory

NINI1988, 2019-03-29:
The checksum in write_memory must be set to `nr_of_bytes ` and not 0xFF.

https://github.com/florisla/stm32loader/blob/1ec166ce22675f263a26df6a5faf1406f5b35ec5/stm32loader/stm32loader.py#L274

Then it's possible to send multiple of 4 bytes and we are not required to send mutliple of 256 bytes.

The following code can be added to the write_memory function to add the padding:
```
if nr_of_bytes % 4 != 0:
    data.extend(([0xFF] * (4 - (nr_of_bytes % 4))))
    nr_of_bytes = len(data)-1
```

florisla, 2019-04-04:
Thanks for reporting this.

AN3155 confirms it; section 3.6 says 'XOR of N and of all data bytes' -- where N is the byte count -1.

I'll support this in a next release, but it may take a while until I resurrect my STM32.

Note: this is an XOR checksum, not a CRC.

nbenm, 2019-04-09:
@NINI1988 can you please put here a diff ?

florisla, 2019-04-09:
@nbenm See this new branch.  Beware, this is untested.

nbenm, 2019-04-09:
@florisla Thanks.
I have cloned 9-write-memory-smaller-byte-count branch.
I'll try it in nearly two weeks after receiving my new devices.
I'll close then the issue if it works fine.

florisla, 2019-04-19:
This is released in v0.4.0.

@nbenm If you're going to test it, could you try the latest version from PyPI?

closed 2019-04-19 by florisla

nbenm, 2019-04-20:
Hi @florisla 
I have downloaded v0.4.0
`Successfully installed progress-1.5 stm32loader-0.4.0`
But I'm still waiting for the devices. The seller told me it will take 2 weeks.

nbenm, 2019-04-27:
I tried the new version.
I didn't have any problems.
That doesn't mean there's none.

florisla, 2019-04-30:
Thanks for reporting!

If anything else is broken I'm sure someone will create an issue soon enough ;-)