florisla/stm32loader #16: Document workflow for contributers

n8henrie, 2019-05-01:
I'm excited to find this fork. I wanted to install into a venv in editable mode, but I'm not terribly familiar with `pyproject.toml` (though I was just looking into it yesterday coincidentally).

```console
$ pip install -e .
...
ERROR: Error installing 'file:///Users/me/stm32loader': editable mode is not supported for pyproject.toml-style projects. pip is processing this project as pyproject.toml-style because it has a pyproject.toml file. Since the project has a setup.py and the pyproject.toml has no "build-backend" key for the "build_system" value, you may pass --no-use-pep517 to opt out of pyproject.toml-style processing. See PEP 517 for details on pyproject.toml-style projects.
```

Is there a reason you only have black settings in it (instead of also including build system info)?

For development / contributors, do you recommend just using `pip install -e . --no-use-pep517`?

florisla, 2019-05-01:
I added `pyproject.toml` solely to configure `black`.  I'm not sure why I haven't seen this issue before, since I use editable install sometimes myself.

I'll add build system metadata to `pyproject.toml` soon; for now you can use `--no-use-pep517`.

Thanks for suggesting contributor documentation, I'll add that too.  The summary is to use `pytest` for unit tests, and `black`/`pylint`/`flake8` for linting. See `noxfile.py` for details.

florisla, 2019-05-02:
This `pip` behaviour is new since pip v19.1, see `https://github.com/pypa/pip/pull/6331`.

Specifying a `build-backend` like this:

```toml
[build-system]
requires = ["setuptools", "wheel"]
build-backend = "setuptools.build_meta"
```

Does not help:

```console
editable mode is not supported for pyproject.toml-style projects. This project is being processed as pyproject.toml-style because it has a pyproject.toml file with a "build-backend" key in the "build_system" value. See PEP 517 for the relevant specification.
```

So for pip >=19.1, you *have* to use `--no-use-pep517` to get an editable install.

n8henrie, 2019-05-06:
Thanks for the quick response -- I didn't realize this was such a new issue. Regardless, looks like it is already fixed in 19.1.1

```console
$ pip freeze --all | grep '^pip'
pip==19.1
$ pip install -e .
Obtaining file:///Users/me/stm32loader
ERROR: Error installing 'file:///Users/me/stm32loader': editable mode is not supported for pyproject.toml-style projects. pip is processing this project as pyproject.toml-style because it has a pyproject.toml file. Since the project has a setup.py and the pyproject.toml has no "build-backend" key for the "build_system" value, you may pass --no-use-pep517 to opt out of pyproject.toml-style processing. See PEP 517 for details on pyproject.toml-style projects.
$ pip install --upgrade pip
Collecting pip
...
Successfully installed pip-19.1.1
$ pip install -e .
Obtaining file:///Users/me/stm32loader
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
    Preparing wheel metadata ... done
Requirement already satisfied: pyserial in ./.venv/lib/python3.7/site-packages (from stm32loader==0.4.1.dev0) (3.4)
Requirement already satisfied: progress in ./.venv/lib/python3.7/site-packages (from stm32loader==0.4.1.dev0) (1.5)
Installing collected packages: stm32loader
  Running setup.py develop for stm32loader
Successfully installed stm32loader
```

closed 2019-05-06 by n8henrie