florisla/stm32loader #18: KeyError length error when reading flash

nbenm, 2019-05-19:
I wanted to download the firmware from a STM32F103
Here's the command I'm using:   
`stm32loader -r -p /dev/cu.usbserial-A5XK3RJT -f F1 dump.bin
`  

And here's the output:  
```
Bootloader version 22
Chip id: 0x410 (STM32F10x Medium-density)
Device UID: FF55-0672-56497751-87184731
Flash size: 64 KiB
Traceback (most recent call last):
  File "/usr/local/bin/stm32loader", line 10, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.7/site-packages/stm32loader/__main__.py", line 37, in main
    stm32loader_main(*sys.argv[1:])
  File "/usr/local/lib/python3.7/site-packages/stm32loader/main.py", line 318, in main
    loader.perform_commands()
  File "/usr/local/lib/python3.7/site-packages/stm32loader/main.py", line 197, in perform_commands
    self.configuration["address"], self.configuration["length"]
KeyError: 'length'
```

Any idea ?

florisla, 2019-05-19:
When reading, you should add a `-l [length]` switch to indicate how many bytes you want to read and also the start address  (`-a [address]`).

E.g. `-l 4096 -a 0x08000000`.

nbenm, 2019-05-19:
Thanks, it works!
Just one more question, if I want all content what should be the best arguments

florisla, 2019-05-19:
The F103 has 64KiB of flash, but unofficially supports 128KiB.

Use `-l 0x10000` or `-l 0x20000`. The address `0x08000000` is already correct.

nbenm, 2019-05-19:
Thanks for all

closed 2019-05-19 by nbenm