florisla/stm32loader #21: Activating read protection

nbenm, 2019-05-26:
Hi,

Is it possible to activate read protection using stm32loader?
I have seen an existing "readout_protect" function in the code. How to use it if possible?
I don't need it to protect any code. I'm doing some tests, and I need to read protect and then unprotect the flash, to see it being erased.
If there's another way to do that, it's ok for me.

florisla, 2019-05-26:
There is no command-line argument to configure read protection currently.

I'm planning to add this, but it will be some weeks before I get to itd

Alternatively, you could write a short Python script and use stm32loader as a library to call `readout_protect`.  I can provide an example script if you want.

nbenm, 2019-05-26:
That would be very nice if you can provide such a script.

nbenm, 2019-05-26:
I have tried this, but I'm not sure it works

```
#!/usr/bin/env python3

from stm32loader.main import Stm32Loader
loader = Stm32Loader()
loader.configuration["port"]="/dev/cu.usbserial-A5XK3RJT"
loader.connect()
loader.stm32.readout_unprotect()
```

nbenm, 2019-05-26:
It works fine switching from readout_unprotect() to readout_protect()
stm32loader "-r" option gives an error
using readout_unprotect() after that makes it possible to read the flash, but it has been erased.
That's exactly what I want.
It would make stm32loader much better if you can add a protect option.

Thanks for this idea you gave me.

closed 2019-05-26 by nbenm

florisla, 2019-05-27:
Thanks for reporting and trying this!