florisla/stm32loader #24: stm32loader STM32F401 not supported?

tomdertech, 2019-07-04:
I am using `STM32F401` which returns a chip ID of `0x433` which is not listed on the ID's within the code. Therefore this is what I get as a response, how do I proceed to use this chip?

```
sudo stm32loader -p /dev/ttyUSB0 -e -w -v -f F4 /home/tom/project.bin
Bootloader version 31
Chip id: 0x433 (Unknown)
Traceback (most recent call last):
  File "/usr/local/bin/stm32loader", line 11, in <module>
    load_entry_point('stm32loader==0.5.0', 'console_scripts', 'stm32loader')()
  File "/home/tom/.local/lib/python3.6/site-packages/stm32loader-0.5.0-py3.6.egg/stm32loader/__main__.py", line 37, in main
  File "/home/tom/.local/lib/python3.6/site-packages/stm32loader-0.5.0-py3.6.egg/stm32loader/main.py", line 317, in main
  File "/home/tom/.local/lib/python3.6/site-packages/stm32loader-0.5.0-py3.6.egg/stm32loader/main.py", line 255, in read_device_details
  File "/home/tom/.local/lib/python3.6/site-packages/stm32loader-0.5.0-py3.6.egg/stm32loader/bootloader.py", line 353, in get_uid
  File "/home/tom/.local/lib/python3.6/site-packages/stm32loader-0.5.0-py3.6.egg/stm32loader/bootloader.py", line 377, in read_memory
  File "/home/tom/.local/lib/python3.6/site-packages/stm32loader-0.5.0-py3.6.egg/stm32loader/bootloader.py", line 635, in _encode_address
struct.error: 'I' format requires 0 <= number <= 4294967295
```
Thanks.
Tom

florisla, 2019-07-10:
Hi,
Pull request #25 adds this chip ID, so the next release will support your `F401`.

In the meantime, you can simply skip the `-f` option altogether; it's not required for writing flash content.

florisla, 2019-08-12:
PR #25 has been merged, so this will be covered in the next release.

tomdertech, 2019-08-14:
I have just pulled the l latest code:

commit 634e05fb1b563137d600d362d8a32d5529a92833 (HEAD -> master, origin/master, origin/HEAD)
Merge: ade9748 a6b166c
Author: Floris Lambrechts <florisla@gmail.com>
Date:   Mon Aug 12 17:13:09 2019 +0200

    Merge branch 'simplify-reset-boot0'

But when I run it still shows `Bootloader version 31` and I still get the problem that I reported above with or without the `-f` flag

florisla, 2019-08-15:
Sorry about that, I'll look into this soon!

tomdertech, 2019-08-21:
Sorry to ask, but do you know when you might be able to fix this? I really would like to use it :-)

florisla, 2019-08-23:
Hi, can you paste the full output of `sudo stm32loader -p /dev/ttyUSB0 -V -e -w -v -f F4 /home/tom/project.bin` (with `-V` for full verbosity)?

Does it mention that Chip ID `0x433` is  `STM32F4xxD/E`?

Reading out the device ID from STM32 `F4`s is peculiar and I don't have a unit myself to test with.

Could you also try branch `24-investigate-f4`?  It will ignore the error and continue normally.
(`git fetch origin` and `git checkout origin/24-investigate-f4`)

tomdertech, 2019-08-27:
Hi,
This log shows the attempt using `-V` before and after I pulled `24-investigate-f4`

tommyers@tommyers-VirtualBox:/stm32loader$ sudo stm32loader -p /dev/ttyUSB0 -V -e -w -v -f F4 /home/tom/project.bin
[sudo] password for tommyers: 
Open port /dev/ttyUSB0, baud 115200
*** Command: Get
    Bootloader version: 0x31
    Available commands: 0x0, 0x1, 0x2, 0x11, 0x21, 0x31, 0x44, 0x63, 0x73, 0x82, 0x92
Bootloader version 31
*** Command: Get ID
Chip id: 0x433 (Unknown)
*** Command: Read memory
Traceback (most recent call last):
  File "/usr/local/bin/stm32loader", line 11, in <module>
    load_entry_point('stm32loader==0.5.0', 'console_scripts', 'stm32loader')()
  File "/home/tommyers/.local/lib/python3.6/site-packages/stm32loader-0.5.0-py3.6.egg/stm32loader/__main__.py", line 37, in main
  File "/home/tommyers/.local/lib/python3.6/site-packages/stm32loader-0.5.0-py3.6.egg/stm32loader/main.py", line 317, in main
  File "/home/tommyers/.local/lib/python3.6/site-packages/stm32loader-0.5.0-py3.6.egg/stm32loader/main.py", line 255, in read_device_details
  File "/home/tommyers/.local/lib/python3.6/site-packages/stm32loader-0.5.0-py3.6.egg/stm32loader/bootloader.py", line 353, in get_uid
  File "/home/tommyers/.local/lib/python3.6/site-packages/stm32loader-0.5.0-py3.6.egg/stm32loader/bootloader.py", line 377, in read_memory
  File "/home/tommyers/.local/lib/python3.6/site-packages/stm32loader-0.5.0-py3.6.egg/stm32loader/bootloader.py", line 635, in _encode_address
struct.error: 'I' format requires 0 <= number <= 4294967295
tommyers@tommyers-VirtualBox:~/stm32loader$ git show --summary
commit ee53162c9ef34e2ba7b285f373817a9bf623ab78 (HEAD -> master, origin/master, origin/HEAD)
Author: Floris Lambrechts <florisla@gmail.com>
Date:   Fri May 3 01:06:04 2019 +0200

    Release: bump version number from v0.5.0-dev to v0.5.0

tommyers@tommyers-VirtualBox:~/stm32loader$ git fetch origin
remote: Enumerating objects: 48, done.
remote: Counting objects: 100% (48/48), done.
remote: Compressing objects: 100% (16/16), done.
remote: Total 59 (delta 33), reused 47 (delta 32), pack-reused 11
Unpacking objects: 100% (59/59), done.
From https://github.com/florisla/stm32loader
   ee53162..634e05f  master            -> origin/master
 * [new branch]      24-investigate-f4 -> origin/24-investigate-f4
 * [new tag]         v0.5.0            -> v0.5.0
tommyers@tommyers-VirtualBox:~/stm32loader$ git checkout origin/24-investigate-f4
Note: checking out 'origin/24-investigate-f4'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b <new-branch-name>

HEAD is now at cadc1fb feat: Ignore any errors during UID lookup
tommyers@tommyers-VirtualBox:~/stm32loader$ git show --summary
commit cadc1fb8a7831dab09e83cea13d46c7c3b56c88e (HEAD, origin/24-investigate-f4)
Author: Floris Lambrechts <florisla@gmail.com>
Date:   Fri Aug 23 17:08:21 2019 +0200

    feat: Ignore any errors during UID lookup

tommyers@tommyers-VirtualBox:~/stm32loader$ sudo stm32loader -p /dev/ttyUSB0 -V -e -w -v -f F4 /home/tom/project.bin
Open port /dev/ttyUSB0, baud 115200
*** Command: Get
    Bootloader version: 0x31
    Available commands: 0x0, 0x1, 0x2, 0x11, 0x21, 0x31, 0x44, 0x63, 0x73, 0x82, 0x92
Bootloader version 31
*** Command: Get ID
Chip id: 0x433 (Unknown)
*** Command: Read memory
Traceback (most recent call last):
  File "/usr/local/bin/stm32loader", line 11, in <module>
    load_entry_point('stm32loader==0.5.0', 'console_scripts', 'stm32loader')()
  File "/home/tommyers/.local/lib/python3.6/site-packages/stm32loader-0.5.0-py3.6.egg/stm32loader/__main__.py", line 37, in main
  File "/home/tommyers/.local/lib/python3.6/site-packages/stm32loader-0.5.0-py3.6.egg/stm32loader/main.py", line 317, in main
  File "/home/tommyers/.local/lib/python3.6/site-packages/stm32loader-0.5.0-py3.6.egg/stm32loader/main.py", line 255, in read_device_details
  File "/home/tommyers/.local/lib/python3.6/site-packages/stm32loader-0.5.0-py3.6.egg/stm32loader/bootloader.py", line 353, in get_uid
  File "/home/tommyers/.local/lib/python3.6/site-packages/stm32loader-0.5.0-py3.6.egg/stm32loader/bootloader.py", line 377, in read_memory
  File "/home/tommyers/.local/lib/python3.6/site-packages/stm32loader-0.5.0-py3.6.egg/stm32loader/bootloader.py", line 635, in _encode_address
struct.error: 'I' format requires 0 <= number <= 4294967295

tomdertech, 2019-08-27:
I tried, `uninstalling` but the package was not recognized, so I tried using the `python` code directly:

```
tommyers@tommyers-VirtualBox:~/stm32loader$ sudo python3 ./stm32loader/__main__.py -p /dev/ttyUSB0 -V -e -w -v -f F4 /home/tom/project.bin
Open port /dev/ttyUSB0, baud 115200
*** Command: Get
    Bootloader version: 0x31
    Available commands: 0x0, 0x1, 0x2, 0x11, 0x21, 0x31, 0x44, 0x63, 0x73, 0x82, 0x92
Bootloader version 31
*** Command: Get ID
Chip id: 0x433 (Unknown)
*** Command: Read memory
Traceback (most recent call last):
  File "./stm32loader/__main__.py", line 41, in <module>
    main()
  File "./stm32loader/__main__.py", line 37, in main
    stm32loader_main(*sys.argv[1:])
  File "/home/tommyers/.local/lib/python3.6/site-packages/stm32loader-0.5.0-py3.6.egg/stm32loader/main.py", line 317, in main
    
  File "/home/tommyers/.local/lib/python3.6/site-packages/stm32loader-0.5.0-py3.6.egg/stm32loader/main.py", line 255, in read_device_details
    if not family:
  File "/home/tommyers/.local/lib/python3.6/site-packages/stm32loader-0.5.0-py3.6.egg/stm32loader/bootloader.py", line 353, in get_uid
    
  File "/home/tommyers/.local/lib/python3.6/site-packages/stm32loader-0.5.0-py3.6.egg/stm32loader/bootloader.py", line 377, in read_memory
    @classmethod
  File "/home/tommyers/.local/lib/python3.6/site-packages/stm32loader-0.5.0-py3.6.egg/stm32loader/bootloader.py", line 635, in _encode_address
    return
struct.error: 'I' format requires 0 <= number <= 4294967295
tommyers@tommyers-VirtualBox:~/stm32loader$ git show --summary
commit cadc1fb8a7831dab09e83cea13d46c7c3b56c88e (HEAD, origin/24-investigate-f4)
Author: Floris Lambrechts <florisla@gmail.com>
Date:   Fri Aug 23 17:08:21 2019 +0200

    feat: Ignore any errors during UID lookup

tommyers@tommyers-VirtualBox:~/stm32loader$ 
```

florisla, 2019-08-27:
It seems your are still using `stm32loader` version `0.5.0` and not the version from git.

I guess you first have to do

    python -m pip uninstall stm32loader

Before it will use the local version.

Or instead of calling `stm32loader`, call `python ./stm32loader/__main__.py`

tomdertech, 2019-08-27:
Forced the `pip uninstall` to work and now setup a `virtualenv`. 

I tried calling the `__main__.py` directly, but now looking for module and not using the python code?
``(env) tommyers@tommyers-VirtualBox:/stm32loader_venv/stm32loader/stm32loader$ sudo python3 __main__.py -p /dev/ttyUSB0 -V -e -w -v -f F4 /home/tom/project.bin 
Traceback (most recent call last):
  File "__main__.py", line 27, in <module>
    from stm32loader.main import main as stm32loader_main
ModuleNotFoundError: No module named 'stm32loader'``

Then tried calling `main.py` but then `bootloader` cannot be found, it seems dependant on the `stm32loader` module?

``(env) tommyers@tommyers-VirtualBox:/stm32loader_venv/stm32loader/stm32loader$ sudo python3 main.py -p /dev/ttyUSB0 -V -e -w -v -f F4 /home/tom/project.bin
Traceback (most recent call last):
  File "main.py", line 30, in <module>
    from . import bootloader
ImportError: cannot import name 'bootloader' ``

florisla, 2019-08-27:
It' still using `v0.5.0`, you can see from the stack trace that the code is inside an `.egg` package in `/home/tommyers/.local/lib/python3.6/site-packages/`.

Even though you start the local `__main__.py`, as soon as an import occurs of `stm32loader` module, it's importing it from the installed site-packages.

Please try to remove `/home/tommyers/.local/lib/python3.6/site-packages/stm32loader-0.5.0-py3.6.egg` altogether.  Else, you'll have to find out why `pip uninstall` refuses to remove `stm32loader`.
Perhaps you have multiple Python installations?

florisla, 2019-08-28:
You have to move one directory higher, otherwise the `stm32loader` module (which is then the current folder) can't be found.

So do

    sudo python3 ./stm32loader/__main__.py

instead of

    sudo python3 ./__main__.py

One other thing you can do inside the `virtualenv` is to install the stm32loader module locally using `pip install -e [path]`:

    python3 -m pip install -e .

tomdertech, 2019-08-28:
Thank you for your help. It appears to now be working :-) I have not tested the actual firmware functions but the `verification` is passed. Will you `commit` this to the `master` branch?

> (env) tommyers@tommyers-VirtualBox:~/stm32loader_venv/stm32loader$ sudo stm32loader -p /dev/ttyUSB0 -V -e -w -v -f F4 /home/tommyers/project.bin
Open port /dev/ttyUSB0, baud 115200
*** Command: Get
    Bootloader version: 0x31
    Available commands: 0x0, 0x1, 0x2, 0x11, 0x21, 0x31, 0x44, 0x63, 0x73, 0x82, 0x92
Bootloader version: 0x31
*** Command: Get ID
Chip id: 0x433 (STM32F4xxD/E)
*** Command: Read memory
Device UID: 003B-0042-30385101-33303837
Flash size: 512 KiB
*** Command: Extended erase memory
Extended erase (0x44), this can take ten seconds or more
    Extended Erase memory done
Write 14 chunks at address 0x8000000...
WritingWrite 256 bytes at 0x8000000
*** Command: Write memory
    [256] bytes to write
    Write memory done
Writing ██∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙ 1/14Write 256 bytes at 0x8000100
*** Command: Write memory
    [256] bytes to write
    Write memory done
Writing ████∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙ 2/14Write 256 bytes at 0x8000200
*** Command: Write memory
    [256] bytes to write
    Write memory done
Writing ██████∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙ 3/14Write 256 bytes at 0x8000300
*** Command: Write memory
    [256] bytes to write
    Write memory done
Writing █████████∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙ 4/14Write 256 bytes at 0x8000400
*** Command: Write memory
    [256] bytes to write
    Write memory done
Writing ███████████∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙ 5/14Write 256 bytes at 0x8000500
*** Command: Write memory
    [256] bytes to write
    Write memory done
Writing █████████████∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙ 6/14Write 256 bytes at 0x8000600
*** Command: Write memory
    [256] bytes to write
    Write memory done
Writing ████████████████∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙ 7/14Write 256 bytes at 0x8000700
*** Command: Write memory
    [256] bytes to write
    Write memory done
Writing ██████████████████∙∙∙∙∙∙∙∙∙∙∙∙∙∙ 8/14Write 256 bytes at 0x8000800
*** Command: Write memory
    [256] bytes to write
    Write memory done
Writing ████████████████████∙∙∙∙∙∙∙∙∙∙∙∙ 9/14Write 256 bytes at 0x8000900
*** Command: Write memory
    [256] bytes to write
    Write memory done
Writing ██████████████████████∙∙∙∙∙∙∙∙∙∙ 10/14Write 256 bytes at 0x8000A00
*** Command: Write memory
    [256] bytes to write
    Write memory done
Writing █████████████████████████∙∙∙∙∙∙∙ 11/14Write 256 bytes at 0x8000B00
*** Command: Write memory
    [256] bytes to write
    Write memory done
Writing ███████████████████████████∙∙∙∙∙ 12/14Write 256 bytes at 0x8000C00
*** Command: Write memory
    [256] bytes to write
    Write memory done
Writing █████████████████████████████∙∙∙ 13/14Write 240 bytes at 0x8000D00
*** Command: Write memory
    [240] bytes to write
    Write memory done
Writing ████████████████████████████████ 14/14
Read 14 chunks at address 0x8000000...
ReadingRead 256 bytes at 0x8000000
*** Command: Read memory
Reading ██∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙ 1/14Read 256 bytes at 0x8000100
*** Command: Read memory
Reading ████∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙ 2/14Read 256 bytes at 0x8000200
*** Command: Read memory
Reading ██████∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙ 3/14Read 256 bytes at 0x8000300
*** Command: Read memory
Reading █████████∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙ 4/14Read 256 bytes at 0x8000400
*** Command: Read memory
Reading ███████████∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙ 5/14Read 256 bytes at 0x8000500
*** Command: Read memory
Reading █████████████∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙ 6/14Read 256 bytes at 0x8000600
*** Command: Read memory
Reading ████████████████∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙ 7/14Read 256 bytes at 0x8000700
*** Command: Read memory
Reading ██████████████████∙∙∙∙∙∙∙∙∙∙∙∙∙∙ 8/14Read 256 bytes at 0x8000800
*** Command: Read memory
Reading ████████████████████∙∙∙∙∙∙∙∙∙∙∙∙ 9/14Read 256 bytes at 0x8000900
*** Command: Read memory
Reading ██████████████████████∙∙∙∙∙∙∙∙∙∙ 10/14Read 256 bytes at 0x8000A00
*** Command: Read memory
Reading █████████████████████████∙∙∙∙∙∙∙ 11/14Read 256 bytes at 0x8000B00
*** Command: Read memory
Reading ███████████████████████████∙∙∙∙∙ 12/14Read 256 bytes at 0x8000C00
*** Command: Read memory
Reading █████████████████████████████∙∙∙ 13/14Read 240 bytes at 0x8000D00
*** Command: Read memory
Reading ████████████████████████████████ 14/14
Verification OK

>

tomdertech, 2019-08-28:
I am confused. It seems every version is 'v0.5.0' when I have `uninstalled` and `reinstalled` that version is always mentioned, regardless of the `branch`. What I think `stm32loader` needs is a `--version` flag to verify what code is actually being executed. I just checked `-h` and no flag exists.

I cannot accept or deny that `master` is truly working for my use case. Since I pulled `#24` with some uninstalling and reinstalling I now have a working use case.

florisla, 2019-08-28:
I think the master branch already works for your case, and the special code in `24-investigate-f4` is not triggered.  At least, I don't see "Error while reading device UID and flash size" in the output.

The reason it did not work before was that you were still using the `v0.5.0` release.

Could you verify if `master` works for you?

    git checkout origin/master
    sudo stm32loader -p /dev/ttyUSB0 -V -e -w -v -f F4 /home/tommyers/project.bin

florisla, 2019-08-28:
You're right, even master branch thinks it's still version 0.5.0.

I'll make sure to add a --version flag in the future.

florisla, 2019-12-31:
Version 0.5.1 is released; please let me know if it does not work correctly.

closed 2019-12-31 by florisla