florisla/stm32loader #26: Missing "timeout" attribute

Maldus512, 2019-07-09:
Hello,
I've been trying to use this tool to flash an stm32f030r8 (NUCLEO-f030r8 demo board) automatically. Unfortunately every attempt ends with the following error:
```
$ stm32loader -p /dev/ttyUSB0 -e -w -v -f F0 rustapp.bin
Bootloader version 31
Chip id: 0x440 (Unknown)
Device UID: UID not supported in this part
Flash size: 64 KiB
Traceback (most recent call last):
  File "/home/maldus/Projects/STM32/rustapp/.env/bin/stm32loader", line 11, in <module>
    load_entry_point('stm32loader==0.5.0', 'console_scripts', 'stm32loader')()
  File "/home/maldus/Projects/STM32/rustapp/.env/lib/python3.7/site-packages/stm32loader-0.5.0-py3.7.egg/stm32loader/__main__.py", line 37, in main
  File "/home/maldus/Projects/STM32/rustapp/.env/lib/python3.7/site-packages/stm32loader-0.5.0-py3.7.egg/stm32loader/main.py", line 318, in main
  File "/home/maldus/Projects/STM32/rustapp/.env/lib/python3.7/site-packages/stm32loader-0.5.0-py3.7.egg/stm32loader/main.py", line 173, in perform_commands
  File "/home/maldus/Projects/STM32/rustapp/.env/lib/python3.7/site-packages/stm32loader-0.5.0-py3.7.egg/stm32loader/bootloader.py", line 426, in erase_memory
  File "/home/maldus/Projects/STM32/rustapp/.env/lib/python3.7/site-packages/stm32loader-0.5.0-py3.7.egg/stm32loader/bootloader.py", line 480, in extended_erase_memory
AttributeError: 'SerialConnection' object has no attribute 'timeout'
```

And indeed it appears the `timeout` attribute referred to in line 480 of `bootloader.py` does not belong to the `SerialConnection` object, but instead to the `serial.Serial` object in its `serial_connection` field. It appears to work when substituting 
```
  previous_timeout_value = self.connection.timeout
```
with
```
  previous_timeout_value = self.connection.serial_connection.timeout
```

Is my assessment correct? Should I submit a pull request to fix this?

byq77, 2019-07-09:
I can confirm. I've encountered the same issue, however I also had problem with option `-f F4` on Nucleo F401RE, it always throws exception. I fixed timeout issue and added additional exception handling for second problem in pull request.

florisla, 2019-07-10:
The fixes by @byq770 will be part of the next release, due in a week.

florisla, 2019-08-12:
Fixes have been merged in master.

closed 2019-08-12 by florisla

Maldus512, 2019-11-22:
I used stm32loader again just now and the same error is still present (albeit at different lines).
Was it re-introduced at some point?

florisla, 2019-12-31:
@Maldus512 Today's version 0.5.1 is the first release which includes these fixes.  Before they were merged, but not released yet.