florisla/stm32loader #38: Missing key in self.configuration, class Stm32Loader, main.py

steverlc, 2020-01-10:
When calling  python main.py -r -p /dev/ttyAMA0 0x02 0x40022080 fl.hex
I encountered 

> File "main.py", line 347, in main
>     loader.perform_commands()
>   File "main.py", line 198, in perform_commands
>     self.configuration["address"], self.configuration["length"]
> KeyError: 'length'

It seems that 'length' is not defined in the dictionary self.configuration[?

florisla, 2020-01-10:
You have to supply the -l argument; there is no 'default' length.

See the example:

    stm32loader -r -p /dev/cu.usbserial-A5XK3RJT -f F1 -l 0x10000 -a 0x08000000 dump.bin 

I'll make sure to produce a nicer error message for missing. Length.

steverlc, 2020-01-10:
Ah! I see now I didn't supply the -l argument, Sorry

PR #40 (2022-02-25): Port argument parsing to argparse
optparse is deprecated, and argparse provides (semi-)automatic help
formatting and much better error messages.

I also added long versions of every option and converted the help
strings to lowercase to conform with the conventions of the options
automatically added by argparse.

Fixes #19.
Fixes #38.
commit 794e8ac5 (2022-02-25): Port argument parsing to argparse

optparse is deprecated, and argparse provides (semi-)automatic help
formatting and much better error messages.

I also added long versions of every option and converted the help
strings to lowercase to conform with the conventions of the options
automatically added by argparse.

Fixes #19.
Fixes #38.
closed 2020-01-10 by steverlc