florisla/stm32loader #41: Flushing via network

asshans, 2020-03-18:
hi.

Why flushing not work via network and socat?

florisla, 2020-03-18:
Please explain how your are using a network.

asshans, 2020-03-18:
Conneсt pl2303 usb-to-rs232 to router, setup access, forward remote port to linux host via socat - `socat PTY,link=/dev/nc0,echo=0,waitslave TCP:10.9.9.98:333`, and try flush - `stm32loader -e -w -v -p /dev/nc0 cmp-wiz.bin`

florisla, 2020-03-18:
I have no experience using socat.

What is the output of the `stm32loader` command?
Also add the `-V` option to make it verbose.

asshans, 2020-03-18:
[root@e7270 cmp-stm32f103]# stm32loader -V -e -w -v -p /dev/nc0 cmp-wiz.bin
Open port /dev/nc0, baud 115200
Activating bootloader (select UART)
Traceback (most recent call last):
  File "/usr/bin/stm32loader", line 11, in <module>
    load_entry_point('stm32loader==0.5.2.dev0', 'console_scripts', 'stm32loader')()
  File "/usr/lib64/python3.8/site-packages/stm32loader-0.5.2.dev0-py3.8.egg/stm32loader/__main__.py", line 37, in main
  File "/usr/lib64/python3.8/site-packages/stm32loader-0.5.2.dev0-py3.8.egg/stm32loader/main.py", line 343, in main
  File "/usr/lib64/python3.8/site-packages/stm32loader-0.5.2.dev0-py3.8.egg/stm32loader/main.py", line 147, in connect
  File "/usr/lib64/python3.8/site-packages/stm32loader-0.5.2.dev0-py3.8.egg/stm32loader/bootloader.py", line 304, in reset_from_system_memory
  File "/usr/lib64/python3.8/site-packages/stm32loader-0.5.2.dev0-py3.8.egg/stm32loader/bootloader.py", line 691, in _enable_boot0
  File "/usr/lib64/python3.8/site-packages/stm32loader-0.5.2.dev0-py3.8.egg/stm32loader/uart.py", line 112, in enable_boot0
  File "/usr/lib64/python3.8/site-packages/pyserial-3.4-py3.8.egg/serial/serialutil.py", line 596, in setRTS
    self.rts = value
  File "/usr/lib64/python3.8/site-packages/pyserial-3.4-py3.8.egg/serial/serialutil.py", line 459, in rts
    self._update_rts_state()
  File "/usr/lib64/python3.8/site-packages/pyserial-3.4-py3.8.egg/serial/serialposix.py", line 629, in _update_rts_state
    fcntl.ioctl(self.fd, TIOCMBIC, TIOCM_RTS_str)
OSError: [Errno 22] Invalid argument
[root@e7270 cmp-stm32f103]#

asshans, 2020-03-18:
[root@e7270 ~]# socat -d -d -d PTY,link=/dev/nc0,echo=0,waitslave TCP:10.9.9.98:333
2020/03/18 23:00:21 socat[11493] I socat by Gerhard Rieger and contributors - see www.dest-unreach.org
2020/03/18 23:00:21 socat[11493] I This product includes software developed by the OpenSSL Project for use in the OpenSSL Toolkit. (http://www.openssl.org/)
2020/03/18 23:00:21 socat[11493] I This product includes software written by Tim Hudson (tjh@cryptsoft.com)
2020/03/18 23:00:21 socat[11493] I setting option "symbolic-link" to "/dev/nc0"
2020/03/18 23:00:21 socat[11493] I setting option "echo" to 0
2020/03/18 23:00:21 socat[11493] I setting option "pty-wait-slave" to 1
2020/03/18 23:00:21 socat[11493] I openpty({5}, {6}, {"/dev/pts/7"},,) -> 0
2020/03/18 23:00:21 socat[11493] N PTY is /dev/pts/7
2020/03/18 23:00:21 socat[11493] I close(6)

2020/03/18 23:01:43 socat[11493] N opening connection to AF=2 10.9.9.98:333
2020/03/18 23:01:43 socat[11493] I starting connect loop
2020/03/18 23:01:43 socat[11493] I socket(2, 1, 6) -> 6
2020/03/18 23:01:43 socat[11493] N successfully connected from local address AF=2 10.9.9.7:60704
2020/03/18 23:01:43 socat[11493] I resolved and opened all sock addresses
2020/03/18 23:01:43 socat[11493] N starting data transfer loop with FDs [5,5] and [6,6]
^C2020/03/18 23:02:19 socat[11493] N socat_signal(): handling signal 2
2020/03/18 23:02:19 socat[11493] N exiting on signal 2
2020/03/18 23:02:19 socat[11493] N socat_signal(): finishing signal 2
2020/03/18 23:02:19 socat[11493] N exit(130)
2020/03/18 23:02:19 socat[11493] I close(5)
2020/03/18 23:02:19 socat[11493] I shutdown(6, 2)

And this is socat log - 23:00:21 is ready
from 23:00:21 - 23:01:43 i try flush, but nothing

23:01:43 - i open port via minicom

florisla, 2022-02-28:
Looking at the stack trace, this crashes when stm32loader is trying to toggle the `BOOT0` pin.

This uses the `RTS` pin of the UART adapter.  It seems your setup does not allow to toggle RTS.

You'll have to control `BOOT0` and `RESET` manually, and remove the functions `enable_reset()` and `enable_boot0()` (in file `uart.py`).

asshans, 2022-03-05:
You can append "-M" option for skip this functions automatic ?

florisla, 2022-03-14:
I can add `--no-rts-dtr`... But first I would like to know if this really works for your use case.

Did you try it already?

closed 2023-10-09 by florisla