florisla/stm32loader #42: Wrong calculation of flash size for F4 controllers

michsens, 2020-06-25:
Hello,

in my tests with an L0 controller I came along an issue according to the calculation of the flash size. 

https://github.com/florisla/stm32loader/blob/497e1f7064129eaabd75b7a73412ac89c1f311a4/stm32loader/bootloader.py#L405

should be changed to:
`flash_size = data[flash_size_lsb_addr] + (data[flash_size_lsb_addr + 1] << 8)`

Only the `data[flash_size_lsb_addr + 1]` should be shifted to the left.

Regards

Robert.

florisla, 2020-06-25:
Obviously, yes.
I guess it was reporting a huuuuge flash size ;-)

Thanks for reporting!

michsens, 2020-06-25:
Your welcome.

It showed somethings about 16 MiB.

florisla, 2022-02-28:
This got fixed in 0bf8d0dbc499b14ef7d37834420bb081bec08fee .

closed 2022-02-28 by florisla