florisla/stm32loader #43: Use flash size address and uid address to find flash size and uid for alternative search (f.i. F4 or L0)

michsens, 2020-06-25:
How about a change of

https://github.com/florisla/stm32loader/blob/497e1f7064129eaabd75b7a73412ac89c1f311a4/stm32loader/bootloader.py#L391-L406

to

```
   def get_flash_size_and_uid(self, device_family): 
        """ 
        Return device_uid and flash_size for F4 family. 
        
        For some reason, F4 and L0 can't read the 12 or 2
        bytes for UID and flash size directly.
        Reading a whole chunk of 256 bytes at 0x1FFFA700 respectivly 0x1FF80000 does work and
        requires some data extraction.
        """
        flash_size_address = int(self.FLASH_SIZE_ADDRESS[device_family])
        uid_address = int(self.UID_ADDRESS.get(device_family, self.UID_ADDRESS_UNKNOWN))
                                                
        data_start_addr = (uid_address & 0xFFFFFF00)
        flash_size_lsb_addr = flash_size_address - data_start_addr
        uid_lsb_addr = uid_address - data_start_addr
        self.debug(0, 'flash_size_address = 0x%X' % flash_size_address)
        self.debug(0, 'uid_address = 0x%X' % uid_address)
        data = self.read_memory(data_start_addr, self.DATA_TRANSFER_SIZE)
        device_uid = data[uid_lsb_addr : uid_lsb_addr + 12]
        flash_size = data[flash_size_lsb_addr] + (data[flash_size_lsb_addr + 1] << 8)
        return flash_size, device_uid
```

Finally

https://github.com/florisla/stm32loader/blob/497e1f7064129eaabd75b7a73412ac89c1f311a4/stm32loader/main.py#L269-L274

Can be changed to

```
            try:
                flash_size = self.stm32.get_flash_size(family)
                device_uid = self.stm32.get_uid(family)
            except:
                # special fix for F4 devices
                flash_size, device_uid = self.stm32.get_flash_size_and_uid(family)
```

Other Controllers can be added.

Regards

Robert.

florisla, 2020-06-25:
I like this.

Can you test which exception occurs in `stm32.get_flash_size(family)` and/or `stm32.get_uid(family)` on F4?

I'd like to catch a specific Exception instead of a broad `except`.

michsens, 2020-06-25:
I'll let you know as soon as I can figure it out.

michsens, 2020-06-25:
Okay, I got it. It is obviously the same as the outer exception:

```
            try:
                flash_size = self.stm32.get_flash_size(family)
                device_uid = self.stm32.get_uid(family)
            except bootloader.CommandError:
                # special fix for F4, L0 and further devices
                flash_size, device_uid = self.stm32.get_flash_size_and_uid(family)
```

michsens, 2020-06-25:
By the way, could you please update following to the bootloader.py to include the L0 family?:

```
CHIP_ID = {
    ...
    # flash size to be looked up  
    0x417: "STM32L05xxx/06xxx",
    ....
    }
```

```
    UID_ADDRESS = {
        ...
        # ST RM0451 25.2 Unique device ID register (96 bits)
        "L0": 0x1FF80050,
        ...
    }
```
and 

```
    FLASH_SIZE_ADDRESS = {
        ...
        # ST RM4510 25.1 Memory size register
        "L0": 0x1FF8007C,
        ...
    }
```

Thanks.

michsens, 2020-07-06:
Hi,

how is the progress? Is there any chance to see the changes is the next release?

Regards

Robert.

florisla, 2020-09-09:
This will be in the next release.

But this is still a while out.

florisla, 2022-02-28:
Merged in db014b2aadd07eabca4e5fac2967bd3495542bdb .

closed 2022-02-28 by florisla