florisla/stm32loader #48: Fails to download if MCU serial port was transmitting before reset

blueskull, 2020-08-25:
As a known behavior of CP2102N, particularly at 115.2kbps, its driver will NOT flush input buffer (device to host) when the port is being opened, thus if the device was transmitting, the data may be still in buffer, causing stm32loader fail to detect ACK.

Testing condition: Ubuntu 20.04, CP2102N, GD32F350G8U6. GigaDevice's official Windows ISP GUI works fine, so it must be the new code, which I just added a UART heartbeat, to be the problem.

Proposed solution: add serial_connection.reset_input_buffer() after reset, or in case CP2102N being very janky, adding the reset command after every write() call.

```
def write(self, *args, **kwargs):
    """Write the given data to the serial connection."""
    ret=self.serial_connection.write(*args, **kwargs)
    self.serial_connection.reset_input_buffer()
    return ret
```

Solution validation: works reliably even if when device sends out data rabidly.

Best Regards

florisla, 2020-09-09:
I'm not sure it's a good idea to reset the buffer so often.

Does the ACK end up in the buffer at all when it's not empty to begin with?

How does this behave if you only reset the buffer at connection time?

blueskull, 2020-09-19:
> How does this behave if you only reset the buffer at connection time?

It works too. I guess it is a better fix.

florisla, 2022-02-28:
Hi @blueskull can you share the code for the fix you are using?

I'd like to incorporate it into the next release.

florisla, 2022-02-28:
Update: I'll incorporate the fix from #46.

florisla, 2022-02-28:
Implemented in c018d70e3498ce4196cd73e592e2a894c00abdb2 .

closed 2022-02-28 by florisla