florisla/stm32loader #49: ModuleNotFoundError: No module named 'stm32loader'

snoop911, 2020-09-05:
Pretty new to this, so most likely something on my end.. but I've installed via :
pip install stm32loader

but in a Python 3.7 x32 shell, I can't seem to import or instantiate it.  It always returns module not found:

import stm32loader    

from stm32loader import __version__, bootloader
from stm32loader.uart import SerialConnection

Alternatively, according to the instructions on pypi.org/project/stm32loader/ it suggests the usage would be something like:
python stm32loader.py -p COM10 -e -w -v somefile.bin

However, I've unzipped "stm32loader-0.5.1.tar.gz" but don't see any stm32loader.py file.

Any pointers what I should try?  
Thanks!

florisla, 2020-09-09:
Hi,

Could you check the version number which is present in file `C:\Python\Python37_x32\lib\site-packages\stm32loader\__init__.py`.
It should read something like

    __version_info__ = (0, 5, 1)

Perhaps your 'pip install' was using another Python version than the one you ended up using.

You can try this:

    py.exe -3.7 -m pip install --upgrade stm32loader
    py.exe -3.7 -c "import stm32loader"

The instructions to use `stm32loader.py` are outdated.  You should either use `stm32loader` as a module, or call the main.py file.

    cd path\to\folder\which-contains-stm32loader-sufolder
    py.exe -3.7 -m stm32loader -p COM10 -e -w -v somefile.bin
    py.exe -3.7 stm32loader/main.py -p COM10 -e -w -v somefile.bin

florisla, 2021-01-20:
Closing in favor of #52.

closed 2021-01-20 by florisla