florisla/stm32loader #54: BlueNRG chip ID is wrong?

rds-tmeyer, 2021-09-06:
Hi,
As per AN4872, the chip ID for BlueNRG chip is 3 bytes :

- The first byte is the "metal fix version", starting at 0x00
- The second byte is the "mask fix version", starting at 0x01
- Third byte most significant nibble is either 0x2 (for BlueNRG-2) or 0x0 (for BlueNRG-1) and the least significant nibble is the flash size : 0x3 for 160kB (BlueNRG-1) and 0xF for 256kB (BlueNRG-2).

Currently I get 0x00012Ffor a BlueNRG-2.

Currently the expected chip ID is 0x11103 which I can't find any source for.

florisla, 2021-10-05:
Thanks for the info!  The code clearly does not support that yet.

I guess we can mask away BYTE1 and BYTE2, and add some BlueNRG-1 and BlueNRG-2 IDs (with various flash sizes) to the list `bootloader.CHIP_IDS`.

florisla, 2023-10-09:
I added the 256kB and 160kB versions of BlueNRG-1 and BlueNRG-2 to master.

closed 2023-10-09 by florisla