florisla/stm32loader #55: Using stm32loader with Jetson Nano

andriskuciks, 2021-10-04:
Hello! 

I was wondering have anyone tried this FW uploading tool on Jetson Nano? I installed stm32loader via pip3 since it supports python3.4 to 3.8 but it seems like there is an architecture issue because when using `stm32loader -h` in terminal it can't find the executable file. Is it possible to use this tool on embedded Linux? 
Or maybe someone has managed to achieve this already on Jetson with another tool? 
I'm using STM32F302VC microcontroller. Another problem probably is that the MCU isn't connected over USB cable but with custom made connector. Connector has all necessary pin connections (USART2 TX/RX, BOOT0, RST pin).

BR, 
Andris

florisla, 2021-10-05:
Stm32loader definitely works fine under Linux, embedded or otherwise.

This looks more like an installation issue rather than an architecture issue.

If `pip3 `works, try this:

`python3 -m stm32loader -h`

What is the port (-p option) you are using?

andriskuciks, 2021-10-05:
Thanks for help, this `python3 -m stm32loader -h` works.

Well since STM32 USART2 is connected to Jetson Nano UART2 for the `-p` argument I'm using port `/dev/ttyTHS2`.

The STM32 device is in bootloader mode, but I still can't manage to flash the firmware. `stm32loader` prints: 
`Activating bootloader (select UART)`
`Bootloader activation timeout -- retrying`
`Can't init into bootloader. Ensure that BOOT0 is enabled and reset the device.`

Full command I'm trying to run: `sudo python3 -m stm32loader -p /dev/ttyTHS2 -e -q firmware.elf`.

Tried it on second (THS1) port and got the same output.

According to STM32F302VC datasheet it should be possible to program MCU via USART2 in bootloader mode.

andriskuciks, 2021-10-06:
Hello again. Today I got my hands on digital analyzer and it turns out that the problem was in my HW setup. Just managed to erase chip memory. Now gonna continue on programming MCU! Thank you for the tool @florisla ! 
BR,
Andris

closed 2021-10-06 by andriskuciks