florisla/stm32loader #56: How to limit flash erased?

tomvarghese, 2021-12-05:
Is there a way to limit how much of the flash is erased.  In my particular case I have some configuration information for the device stored at the end of flash and my program is much smaller than the full flash.  

I would prefer not to lose the configuration when I update the firmware.

Anyone done a fork of this that has addressed this?

I'm using the STM32L0xx processors  only at the moment.

florisla, 2022-01-29:
The command line does not offer this option currently.

But the erase_memory() method does allow you to specify the addresses of individual pages which need to be erased.

florisla, 2023-10-10:
The next release will support `--erase` in combination with `--address` and `--length` like this:

    stm32loader --erase --address 0x08000000 --length 0x2000 --port /dev/cu.usbserial-A5XK3RJT

closed 2023-10-10 by florisla