florisla/stm32loader #57: Bootloader hex file for STM32F103xx

embpic, 2022-01-29:
Hi,
I am using STM32F103xx and i wanted to use the UART bootlaoder. Where can i get hex file for UART bootloader so that i can use stm32loader python script.
Is this work with default UART bootloader ?

florisla, 2022-01-29:
It looks like `BOOT0` is set correctly for an STM32F103.  You need to press the _Reset_ button on the blue STM32F1 board before starting `stm32loader`.

If that does not help, please consider:
* You can use the UART adapter's `DTR` pin to toggle `RESET` automatically.
* Check if both devices remain powered (e.g. place a LED with a resistor on a 5V or 3.3V line).
* Check if the UART adapter uses 5V or 3.3V levels, and if that is compatible with your target board.
* Re-do all wire connections.  Wires can be flaky!

florisla, 2022-01-29:
Also, you can lower the baudrate and double-check that the UART adapter has pull-up resistors on the RX and TX lines,

closed 2023-10-09 by florisla