florisla/stm32loader #76: [Request] Please test the 'device-table' branch against your devices!

florisla, 2024-01-24:
Hi all,

This partial rewrite of stm32loader allows usage without supplying the `--family` argument.

Since this is a fairly impactfull change, I hesitate to release it without proper testing on at least some device families.

So please run this 'experimental' stm32loader on your device and report back with success or errors.

The branch:
https://github.com/florisla/stm32loader/tree/feat/device-table

To install:

```
pip install git+https://github.com/florisla/stm32loader.git@feat/device-table
```

dbeinder, 2024-01-24:
The branch doesn't work yet, because `stm32.detect_device()` is called only in the test but not in main and `self.device` never gets set. I fixed it in my PR but forgot mentioning it.

florisla, 2024-01-24:
#74 Tested BlueNRG .

florisla, 2024-02-19:
Thanks for testing!

The next release will be based on the device-table branch so it seems no change is needed there.

florisla, 2024-02-19:
#75 tested `STM32C0`.

closed 2025-01-10 by florisla