florisla/stm32loader #78: Implement Go command

andrewteta, 2024-06-17:
I would like to be able to execute the program after flashing without power cycling the device. I believe this is possible using the "go" command.

florisla, 2024-06-18:
The Go command is already implemented internally, it's only the command line options which is missing.

Does it make sense to split this up into two invocations?

    stm32loader --write firmware.bin
    stm32loader --go 0x08000000

dbeinder, 2024-06-18:
I'd prefer if it was possible to add `--go` as an extra step to a flash call

florisla, 2025-01-10:
Coming back to this, I agree we should add the `--go` argument and it makes sense to execute it as the last step, after erase and write.

florisla, 2026-10-28:
On closer look, `-g ADDRESS` (or `--go-address ADDRESS` is already available as an option.  It can be combined with --write, --verify, ...   I will test if it still works.

Do let me know if it does not work for you.

closed 2026-10-28 by florisla